Rotrics-Dev/rotrics-studio-app-issue #1: Development plan

liumingzw, 2020-07-13:
Hi，everyone：
I am Walker，the maintainer and main developer of Rotrics Studio.
The following is my development plan.

- [ ] Main feature: 3d print, laser, write and draw, basic control, firmware upgrade and etc.

- [ ]  Multi language support.  The feature will be completed at 2020.07.30

- [ ] Linux support. The feature will be completed at 2020.08.30

bmagyar, 2020-08-10:
Very happy to help out testing the Linux version, just let me know!

Ice-Panda, 2020-08-27:
First of all, I'd like to thank the author for creating such an amazing software platform.
Also as a developer, I would like to ask you what kind of programming language is used to write such beautiful and practical software.
By the way, I have also seen your zhihu article and hope to learn from you.

DavidVives, 2020-09-01:
Great work! let us know on how can we help improve !

liumingzw, 2020-10-01:
> First of all, I'd like to thank the author for creating such an amazing software platform.
> Also as a developer, I would like to ask you what kind of programming language is used to write such [...]

The main programming language is javascript. And a little glsl written by c.
In the future, webassembly(c/c++ or rust) will be added to improve performance.

The main frameworks or module are React, Redux, Antd, Threejs, Electron, etc..

joaozorro, 2020-12-30:
Hello, great work! Do you have an updated release date for the "Enhance scratch part to control camera."? I'm waiting for my camera to arrive but without support in Scratch it will be difficult to test it.
Thanks