Rotrics-Dev/rotrics-studio-app-issue #4: Laser Engraving/Cutting settings

Meatstake, 2020-08-29:
I think, the laser engraving expierence could be augmented with further features:
- Greyscaling not only with the filling pattern, but also with "Laser Power" and/or "Burn Time" , at least as an option. This also would alow direct "dot to pixel" pattern for engraving in "grayscale" and the engraved images would not look like a labyrinth-like offset print from an old newspaper, as we have it right now.
- B&W engraving without interlase (actual lineal mode), or also with "dot method" as on other Laser Engravers.
- B&W engraving without string/interlacing/breaks also for non SVG Files. Just like Rotrics does it right now with a pen in "wrighting mode".
- Layers support with dedicated setting for each layer in the App to engrave and cut the engraved out and so on.

Master-68, 2020-09-27:
Yes, please!

Master-68, 2020-09-27:
Another thing: When selecting multi-pass laser, the z value it lowers the arm cannot be set to less than 1 mm. This restriction should be removed.

MichaelDu9226, 2020-10-14:
> Another thing: When selecting multi-pass laser, the z value it lowers the arm cannot be set to less than 1 mm. This restriction should be removed.
Restriction has been removed.