Rotrics-Dev/rotrics-studio-app-issue #12: add M125 "parking position"

ikkez, 2020-09-06:
When the robot is at home or below and we turn it off, it might occur that the robot arm just falls down on the table. 
It would be useful to have a button or a gcode to let the arm park in it's real starting position, where it is save to be turned off.

- [ ] implement gcode for M125, so the robot can go back to it's parking position at real 0,0,0

Master-68, 2020-09-27:
Yes, definitely!

MichaelDu9226, 2020-10-14:
Thanks for your advice. We will implement this function in the firmware as soon as possible.  Then add to Rotrics Studio.

MichaelDu9226, 2020-10-23:
@ChrisWhite1985 @Master-68 @ikkez We are considering add M125 to Dexarm. But which position is the best position? Especially after 3D printing is done.

Master-68, 2020-10-23:
From my point of view there are 3 reasons why there should be such a position:

1.) the arm should not drop down to the surface when power is switched off
2.) after a 3d print or a laser engraving has finished, the arm should be out of the way to easily access the printed / engraved model
3.) there should be a convenient way to move the arm to a safe "storage position"

So from my point of view, a good position would be the one so that it fits into the storage box. X = 0, y at reachable minimum, z accordingly so that y minimum can be reached.
To make sure that the arm does not collide with a printed model while moving to that position, the movement order should be to move to the minimum reachable y at current height first, and then move to target z while keeping y to the reachable minimum at all time while moving z up or down.