Rotrics-Dev/rotrics-studio-app-issue #13: Laser, Draw: Object Transformation

ikkez, 2020-09-06:
- [ ] when you click on an object, make it possible to move it via keyboard arrow keys
- [ ] when you change values in the transformation properties, the changes are not visible instantly in the preview workarea. You currently need to switch between the fields or hit enter all the time and click again to see the updates. Update the workarea directly when changing values via keyboard, or after typing (like after 100ms timeout)

Master-68, 2020-09-27:
Useful, but only if the auto-update can be done in a reasonable time. If it take several seconds to create the new view of the object, this would only break the workflow. In that case, an "Update View" button would be better.

MichaelDu9226, 2020-10-14:
Thanks for your advice.  We will consider this feature.