Rotrics-Dev/rotrics-studio-app-issue #7: head position on screen

ikkez, 2020-09-06:
Currently the center (home) is at 300mm Y. 

[image]

It would be great to see that in Laser-Mode so, you can place the things to engrave more accurately.

please add:

- [ ] display current head postion on the screen in Basic, Laser and Draw Mode
- [ ] add a button to enable movement mode, so when you click somewhere in the work area, the robot head moves to the clicked position

Master-68, 2020-09-27:
Also a "run outline" button (a square, that shows the max dimensions of the picture would do) with the laser "on" at extremely low power (much less that 1%) would be useful, to place the object at the right position would be helpful. Its often difficult to guess, where the engraved picture will really be.

eleven29, 2020-09-28:
> Also a "run outline" button (a square, that shows the max dimensions of the picture would do) with the laser "on" at extremely low power (much less that 1%) would be useful, to place the object at [...]

I think this warrants to be its own issue as it's separate from seeing the position of the head on-screen.

MichaelDu9226, 2020-10-14:
Thanks for your advice. I've fed back to the software engineer. Maybe the next version can solve it