Rotrics-Dev/rotrics-studio-app-issue #15: some buttons to be able to create our own macros

MichaelDu9226, 2020-09-27:
Some buttons to be able to create our own macros (GCode) will do wonders to improve functionality.

Master-68, 2020-09-27:
Very useful! This would make things much more comfortable! Best would be, if these buttons were visible all the time, regardless in which screen you are. for example in a detachable dock or a free movable extra window. A minimum of 10 or 12 (3x4 for example?) buttons should be available. One should be able to give each button an own name and the button should be able to send multi-line gcode to the arm, when pressed (not only one gcode).
A nice feature (though I understand, that this would be difficult to implement) would be, if these buttons would support some kind of logic capability, so one could for example create things like: Send gcode to heat up nozze; wait until temperature x is reached; move to work origin; ... etc.

llepe, 2020-09-28:
Definitely support @Master-68 many errors could be "patched" by us and you could get many ideas on what do we need by the sharing of the "scripts"