Rotrics-Dev/rotrics-studio-app-issue #22: Incorrect generation G-Code when drawing a vector image

0x600, 2022-04-23:
OS Version: Windows 10 Pro 21H2 19044.1645
Rotrisk Studio version: 1.0.2
Rotkris Dex ARM firmware version: 3.3.2

Description of the problem:
After selecting a vector image, the robot only draws a straight line. It looks like the G-Code is being created incorrectly. However, with other operations, such as a laser, the G-Code is created correctly. 
UPD: Also, when programming in Python, there is no such problem.

NEUROMAX777, 2024-04-26:
Аналогичная проблема! скину сюда текст G-code
;TOOL_PATH_RENDER_METHOD_LINE
;----------- Start Gcode -----------
M2000;custom:line mode
M888 P0;custom:header is write&draw
;-----------------------------------
G0 Z5
G0 F2000
G1 F2000
G0 E-42.06 Y213.37
G1 Z0
G1 E-46.87 Y213.37
G1 E-46.87 Y228.87
G1 E-38.89 Y228.87
G1 E-37.84 Y228.85
G1 E-36.14 Y228.66
G1 E-35.49 Y228.49
G1 E-34.93 Y228.26
G1 E-33.99 Y227.55
G1 E-33.60 Y227.08
G1 E-33.28 Y226.54
G1 E-32.92 Y225.28
G1 E-32.88 Y224.57
G1 E-32.88 Y224.57
G1 E-32.91 Y223.95
G1 E-33.19 Y222.84
G1 E-33.43 Y222.35
G1 E-33.73 Y221.90
G1 E-34.49 Y221.14
G1 E-34.94 Y220.83
G1 E-34.94 Y220.83
G1 E-35.67 Y220.49
G1 E-36.62 Y220.21
G1 E-36.62 Y220.21
G1 E-35.88 Y219.93
G1 E-35.37 Y219.64
G1 E-35.37 Y219.64
G1 E-35.05 Y219.34
G1 E-34.61 Y218.83
G1 E-34.19 Y218.28
G1 E-33.94 Y217.87
G1 E-33.94 Y217.87
G1 E-31.62 Y213.37
G1 E-37.04 Y213.37
G1 E-39.59 Y218.11
G1 E-40.06 Y218.87
G1 E-40.46 Y219.31
G1 E-40.46 Y219.31
G1 E-41.01 Y219.57
G1 E-41.64 Y219.66
G1 E-41.64 Y219.66
G1 E-42.06 Y219.66
G1 E-42.06 Y213.37
G0 Z5
G0 E-42.06 Y225.74
G1 Z0
G1 E-42.06 Y222.59
G1 E-40.04 Y222.59
G1 E-39.56 Y222.65
G1 E-38.77 Y222.81
G1 E-38.77 Y222.81
G1 E-38.34 Y222.97
G1 E-37.99 Y223.29
G1 E-37.77 Y223.71
G1 E-37.69 Y224.19
G1 E-37.69 Y224.19
G1 E-37.72 Y224.54
G1 E-37.96 Y225.12
G1 E-38.17 Y225.34
G1 E-38.46 Y225.52
[... 312 lines cut ...]
G1 E48.01 Y218.78
G1 E48.25 Y218.36
G1 E48.53 Y217.48
G1 E48.56 Y217.01
G1 E48.56 Y217.01
G1 E48.53 Y216.53
G1 E48.23 Y215.58
G1 E47.98 Y215.12
G1 E47.64 Y214.68
G1 E46.71 Y213.95
G1 E46.12 Y213.65
G1 E45.44 Y213.41
G1 E43.72 Y213.14
G1 E42.69 Y213.11
G1 E42.69 Y213.11
G1 E41.26 Y213.16
G1 E39.07 Y213.60
G1 E38.30 Y213.99
G1 E37.71 Y214.47
G1 E36.87 Y215.72
G1 E36.63 Y216.48
G0 Z5
G0 E1.48 Y220.89
;----------- End Gcode -------------
;-----------------------------------

NEUROMAX777, 2024-04-26:
Где тут ошибка кода? Рисует прямую линию!

InnovusSollertia, 2024-04-26:
> Где тут ошибка кода? Рисует прямую линию!

Так а должен рисовать векторное изображение

kreckergay, 2024-08-28:
same problem. выдаёт линию и при рисовании, и лазером. 3d не тестил. файл аналогичен скинутому. есть подвижки?

InnovusSollertia, 2024-08-28:
> same problem. выдаёт линию и при рисовании, и лазером. 3d не тестил. файл аналогичен скинутому. есть подвижки?

Нет. Компания забросила этот продукт и больше не развивает. Думаю насчет реверс-инжиниринга и написания нормальной прошивки, но это пока что мечты...

closed 2025-11-05 by InnovusSollertia